MazamaScience/AirSensor #162: Creating PAT when duplicate sensor names exist

BrandonAQ, 2019-10-09:
Creating PAT data objects for the Seal Beach community. I am running into an issue where pat_createNew fails to create because two sensors share the same label.. This occurs where a PA-II sensor has been replaced with a new sensor that is labeled identical to prior sensor. This is the case for SCSB_09, SCSB_24, and SCSB 32. As a result, the PAT is not created. 

Using AirSensor 0.5.5
library(AirSensor)
library(stringr)
library(rlang)
setArchiveBaseUrl("http://smoke.mazamascience.com/data/PurpleAir")
# loads a version of the pas that includes every sensor installed (i.e., inactive ones)
pas <- pas_load(archival = TRUE)

# Create PAT data object
pat_scsb_09 <- pat_createNew(pas = pas, label = "SCSB_09", startdate = 20191001, enddate = 20191008, timezone = NULL, baseURL = "https://api.thingspeak.com/channels/")

Error Message:
"Error in pat_createNew(pas = pas, label = "SCSB_09", startdate = 20191001,  : 
  2 senors share the label 'SCSB_09'"

Looking for options to create pat for replaced sensor. If loading pas with archive = false, then I can create a PAT for the sensor currently active, if that unit is still active.

jonathancallahan, 2019-10-10:
Resolved in version 0.5.6.

```
> pat_scsb_09 <- pat_createNew(pas = pas, label = "SCSB_09", startdate = 20190101, enddate = 20190301)
Error in pat_createNew(pas = pas, label = "SCSB_09", startdate = 20190101,  : 
  Multilpe sensors share this label.You must specify the 'id' parameter as one of: '20071, 2016'
```

Decent error message. Ok. Include the `id` for the more recent installation:

```
> pat_scsb_09 <- pat_createNew(pas = pas, label = "SCSB_09", startdate = 20190101, enddate = 20190301, id = "20071")
```

jonathancallahan, 2019-10-10:
..

closed 2019-10-10 by jonathancallahan