MazamaScience/AirSensor #163: possible issue with "archival" option in pas_load()

acollier-oxandale, 2019-10-15:
There appears to be a problem with the "archival" option for pas_load()

Running the following code, using 

1. pas <- pas_load()
2. pas <- pas_load(archival = TRUE)
3. pas <- pas_load(archival = FALSE) 

all yield the same result - 25 sensors for the SCAP community, however, last week running this same code with "archival  = TRUE" resulted in 41 sensors.

_______________________________________________________________________________

Code:

library(AirSensor)
library(stringr)
library(rlang)
library(openair)

setArchiveBaseUrl("http://smoke.mazamascience.com/data/PurpleAir")

pas <- pas_load(archival = TRUE)

scap <- pas %>%
   pas_filter(str_detect(label, "^[Ss][Cc][Aa][Pp]_"))
sens <- scap$label
sens_a <- sens[which(str_detect(sens, " B") == FALSE)]

This weeks output:

> sens_a
 [1] "SCAP_01" "SCAP_03" "SCAP_05" "SCAP_07" "SCAP_10" "SCAP_14" "SCAP_17"
 [8] "SCAP_19" "scap_26" "SCAP_28" "SCAP_31" "SCAP_34" "SCAP_35" "SCAP_36"
[15] "SCAP_41" "SCAP_46" "SCAP_49" "SCAP_50" "SCAP_52" "SCAP_55" "SCAP_58"
[22] "SCAP_5G" "SCAP_68" "SCAP_77" "SCAP_87"

Last weeks output:

> sens_a
 [1] "SCAP_01" "SCAP_03" "SCAP_04" "SCAP_05" "SCAP_07" "SCAP_10" "SCAP_13"
 [8] "SCAP_14" "SCAP_15" "SCAP_16" "SCAP_17" "SCAP_18" "SCAP_19" "SCAP_23"
[15] "SCAP_24" "scap_26" "SCAP_27" "SCAP_28" "SCAP_29" "SCAP_31" "SCAP_33"
[22] "SCAP_34" "SCAP_35" "SCAP_36" "SCAP_37" "SCAP_41" "SCAP_42" "SCAP_43"
[29] "SCAP_46" "SCAP_48" "SCAP_49" "SCAP_50" "SCAP_51" "SCAP_52" "SCAP_53"
[36] "SCAP_55" "SCAP_58" "SCAP_5G" "SCAP_68" "SCAP_77" "SCAP_87"

jonathancallahan, 2019-10-15:
Running the following code I see that something changed on Oct 14. Now checking to see whether we did anything or it is a change in what PurpleAir provides.

```
> library(dplyr)
> 
> for ( datestamp in 20191010:20191015) {
+   
+   cat(paste0("\nUsing pas_", datestamp, "\n"))
+   
+   labels <- pas_load(datestamp, archival = TRUE) %>% 
+     pas_filter(str_detect(label, "^[Ss][Cc][Aa][Pp]_")) %>%
+     pas_filter(!stringr::str_detect(label, " ")) %>% 
+     pull(label)
+   
+   print(labels)
+   
+ }

Using pas_20191010
 [1] "SCAP_01" "SCAP_03" "SCAP_04" "SCAP_05" "SCAP_07" "SCAP_10" "SCAP_13" "SCAP_14" "SCAP_15" "SCAP_16" "SCAP_17" "SCAP_18" "SCAP_19" "SCAP_23" "SCAP_24" "scap_26"
[17] "SCAP_27" "SCAP_28" "SCAP_29" "SCAP_31" "SCAP_33" "SCAP_34" "SCAP_35" "SCAP_36" "SCAP_37" "SCAP_41" "SCAP_42" "SCAP_43" "SCAP_46" "SCAP_48" "SCAP_49" "SCAP_50"
[33] "SCAP_51" "SCAP_52" "SCAP_53" "SCAP_55" "SCAP_58" "SCAP_5G" "SCAP_68" "SCAP_77" "SCAP_87"

Using pas_20191011
 [1] "SCAP_01" "SCAP_03" "SCAP_04" "SCAP_05" "SCAP_07" "SCAP_10" "SCAP_13" "SCAP_14" "SCAP_15" "SCAP_16" "SCAP_17" "SCAP_18" "SCAP_19" "SCAP_23" "SCAP_24" "scap_26"
[17] "SCAP_27" "SCAP_28" "SCAP_29" "SCAP_31" "SCAP_33" "SCAP_34" "SCAP_35" "SCAP_36" "SCAP_37" "SCAP_41" "SCAP_42" "SCAP_43" "SCAP_46" "SCAP_48" "SCAP_49" "SCAP_50"
[33] "SCAP_51" "SCAP_52" "SCAP_53" "SCAP_55" "SCAP_58" "SCAP_5G" "SCAP_68" "SCAP_77" "SCAP_87"

Using pas_20191012
 [1] "SCAP_01" "SCAP_03" "SCAP_04" "SCAP_05" "SCAP_07" "SCAP_10" "SCAP_13" "SCAP_14" "SCAP_15" "SCAP_16" "SCAP_17" "SCAP_18" "SCAP_19" "SCAP_23" "SCAP_24" "scap_26"
[17] "SCAP_27" "SCAP_28" "SCAP_29" "SCAP_31" "SCAP_33" "SCAP_34" "SCAP_35" "SCAP_36" "SCAP_37" "SCAP_41" "SCAP_42" "SCAP_43" "SCAP_46" "SCAP_48" "SCAP_49" "SCAP_50"
[33] "SCAP_51" "SCAP_52" "SCAP_53" "SCAP_55" "SCAP_58" "SCAP_5G" "SCAP_68" "SCAP_77" "SCAP_87"

Using pas_20191013
 [1] "SCAP_01" "SCAP_03" "SCAP_04" "SCAP_05" "SCAP_07" "SCAP_10" "SCAP_13" "SCAP_14" "SCAP_15" "SCAP_16" "SCAP_17" "SCAP_18" "SCAP_19" "SCAP_23" "SCAP_24" "scap_26"
[17] "SCAP_27" "SCAP_28" "SCAP_29" "SCAP_31" "SCAP_33" "SCAP_34" "SCAP_35" "SCAP_36" "SCAP_37" "SCAP_41" "SCAP_42" "SCAP_43" "SCAP_46" "SCAP_48" "SCAP_49" "SCAP_50"
[33] "SCAP_51" "SCAP_52" "SCAP_53" "SCAP_55" "SCAP_58" "SCAP_5G" "SCAP_68" "SCAP_77" "SCAP_87"

Using pas_20191014
 [1] "SCAP_01" "SCAP_03" "SCAP_05" "SCAP_07" "SCAP_10" "SCAP_14" "SCAP_17" "SCAP_19" "scap_26" "SCAP_28" "SCAP_31" "SCAP_34" "SCAP_35" "SCAP_36" "SCAP_41" "SCAP_46"
[17] "SCAP_49" "SCAP_50" "SCAP_52" "SCAP_55" "SCAP_58" "SCAP_5G" "SCAP_68" "SCAP_77" "SCAP_87"

Using pas_20191015
 [1] "SCAP_01" "SCAP_03" "SCAP_05" "SCAP_07" "SCAP_10" "SCAP_14" "SCAP_17" "SCAP_19" "scap_26" "SCAP_28" "SCAP_31" "SCAP_34" "SCAP_35" "SCAP_36" "SCAP_41" "SCAP_46"
[17] "SCAP_49" "SCAP_50" "SCAP_52" "SCAP_55" "SCAP_58" "SCAP_5G" "SCAP_68" "SCAP_77" "SCAP_87"
```

jonathancallahan, 2019-10-15:
Yep. As demonstrated below, it looks like Purple Air has dropped a lot of sensors from their recent json. Presumably they have purged sensors that haven't been reporting for a while. Their focus is on real-time.

This is not something that is fixable other than to suggest that you load older PAS files when working with archival data.

Here is code you can run to see that their current json doesn't include the sensors you are looking for:

```
# ----- Download raw data ----------------------------------------------------

# NOTE:  using Hadley Wickham style: 
# NOTE:  https://github.com/hadley/httr/blob/master/vignettes/quickstart.Rmd
r <- httr::GET("https://www.purpleair.com/json")

if ( httr::http_error(r) ) {
  stop("Download failed")
} else {
  content <- httr::content(r, as="text", encoding="UTF-8") # don't interpret
}

# Convert JSON to an R list
PAList <- jsonlite::fromJSON(
  content,
  simplifyVector=TRUE,
  simplifyDataFrame=TRUE,
  simplifyMatrix=TRUE,
  flatten=FALSE
)

# > names(PAList)
# [1] "mapVersion"       "baseVersion"      "mapVersionString" "results"

# Pull out the dataframe of results
resultsDF <- PAList$results

# Original json calls it "Label" with a capital "L"
resultsDF %>%
  filter(str_detect(Label, "^[Ss][Cc][Aa][Pp]_")) %>%
  filter(!stringr::str_detect(Label, " B")) %>% 
  pull(Label)
```

acollier-oxandale, 2019-10-15:
Thank you so much for looking into and clarifying that, it will be important to know as we work through our analysis.  Also, I did just try loading a historic pas file with the archival option set to TRUE and it worked as expected. 

Thanks again!

jonathancallahan, 2019-10-15:
Over the years I've learned that web resources sometimes change or go away unexpectedly. So if you are depending on something you get from the web you should probably download and save it somewhere.

Then, when building package functionality, always make sure it is easy to load an older version.

You're welcome!

closed 2019-11-01 by jonathancallahan