MazamaScience/AirSensor #165: check on local time in all SoH_ functions

jonathancallahan, 2019-10-16:
All of the state-of-health functions should:

* trim any incoming time series to local time day boundaries
* calculate daily metrics based on local time

kayleighwilson, 2019-10-23:
This is currently in an odd state. Both `SoH_dailyPCTValid` and `SoH_dailyPCTReporting` call `pat_aggregateOutlierCounts` for the time period of "1 day" which appears to be doing strange magic with the timezones.

kayleighwilson, 2019-10-23:
Update: I reduced the aggregation period in `pat_aggregateOutlierCounts` to 1 hour and used dplyr tools to do additional aggregation to avoid timezone issues. There still might be a better way to work out the problem but for now, this is the functioning work around

closed 2019-11-01 by jonathancallahan