MazamaScience/AirSensor #177: hourly_AB_01 documentation in R-package

BrandonAQ, 2019-11-04:
Would it be more accurate to say: 
3. Invalidate data where: **(pm25 < 100)** & (p-value < 1e-4) & (mean_diff > 10)
4. Invalidate data where: (pm25 **>** 100) & (mean_diff > 20)

[image]

jonathancallahan, 2019-11-04:
I don't think the phrase "more accurate" applies here with the statistical meaning of "accurate" as in "accuracy vs precision".

But, whatever phrase we use, there is room for improving the QC algorithm. This is also why the original algorithm is `AB_01` so that we have 98 more tries to get it right!

Any  QC algorithm is an attempt to find the right balance between type 1 and type 2 errors, while trying to minimize both. Given how messy the data are, this can be challenging.

I would agree that the `AB_01` algorithm is a little unforgiving of sensors whose A and B channels both seem to be tracking increasing levels of pm25 but whose values differ by a significant amount.

The suggested improvement might address this but may also fail to flag some sensor failure modes. The best approach would be to test the algorithm against a set of sensor time series that reflect different conditions:

- performing well, clean air
- performing well, urban pollution
- performing well, urban July 4-6
- performing well, smoke intrusion
- sensor failure mode A
- sensor failure mode B
- sensor failure mode C
- sensor failure mode D

The failure modes vignette has some suggestions:

https://mazamascience.github.io/AirSensor/articles/purpleair_failure_modes.html

**Testing new QC functions**

The package was written with novel QC algorithms in mind. The `pat_createAirSensor()` function that converts a raw PAT object into an hourly AirSensor object has the following function signature:

```
  pat = NULL,
  period = "1 hour",
  parameter = "pm25",
  channel = "ab",
  qc_algorithm = "hourly_AB_01",
  min_count = 20,
  aggregation_FUN = pat_aggregate
```

A user can test a new QC algorithm without rebuilding the package by copying `PurpleAirQC_hourly_AB_01.R` from the web and modifying the name and the algorithm. Then just source the new function in RStudio and run `pat_createAirSensor()` specifying the new algorithm to see if the result looks better than with `hourly_AB_01`.

**NOTE:**

If a decision were made to replace the existing QC algorithm with a new one, it would be responsible to completely rebuild the entire `airsensor` portion of the database so that those datasets all reflect the same treatment.

closed 2019-11-04 by jonathancallahan

BrandonAQ, 2019-11-04:
Jon, 

Thanks for the in depth look at this, but you missed the title of issue / point. 
I am not asking for a change to QC_01, but asking if the documentation accompanying QC_01 is "accurate" or "correct". This has to to do with documentation accompanying the R-package. My version is 0.5.6 if that help. The picture attached is what shows up. 
 
Looking at documentation alone, does it accurately describe the function: 
1. If in number 3 you are invalidating all A/B with  mean_diff > 10, then what is purpose of #4. That is why I suggest the documentation may be missing the (pm25 < 100) for mean difference. Maybe not, let me know..... 
2. Look at the sign on #4 for PM2.5. Is this correct? Should it be > or <? 

Thanks, 
B

BrandonAQ, 2019-11-04:
When reviewing documentation, I do note that I should have reviewed code prior to submitting this issue to determine if description is appropriate. The code - inserted below- does appear to match description.... but seems now there may be issue with intent of function vs actual function. 

Note comment: 
    # A difference of 20 ug/m3 should only be seen at very high levels.
    # Invalidate data where:  (mean < 100) & (mean_diff > 20)
   

MazamaCoreUtils::stopIfNull(aggregationStats)
  
  # ----- hourly_AB_01 ---------------------------------------------------------
  
  hourlyData <-
    aggregationStats %>%
    # Create pm25 by averaging the A and B channel aggregation means
    dplyr::mutate(pm25 = (.data$pm25_A_mean + .data$pm25_B_mean) / 2) %>%
    # Calculate min_count and mean_diff for use in QC
    dplyr::mutate(min_count = pmin(.data$pm25_A_count, .data$pm25_B_count, na.rm = TRUE)) %>%
    dplyr::mutate(mean_diff = abs(.data$pm25_A_mean - .data$pm25_B_mean)) %>%
    # hourly_AB_01 follows
    # When only a fraction of the data are reporting, something is wrong.
    # Invalidate data where:  (min_count < SOME_THRESHOLD)
    dplyr::mutate(pm25 = replace(
      .data$pm25, 
      which(.data$min_count < min_count), 
      NA) 
    ) %>%
    # When the means are significantly different AND 'large', something is wrong.
    # Invalidate data where:  (p-value < 1e-4) & (mean_diff > 10)
    dplyr::mutate(pm25 = replace(
      .data$pm25,
      which( (.data$pm25_p < 1e-4) & (.data$mean_diff > 10) ),
      NA)
    ) %>% 
    # A difference of 20 ug/m3 should only be seen at very high levels.
    # Invalidate data where:  (mean < 100) & (mean_diff > 20)
    dplyr::mutate(pm25 = replace(
      .data$pm25,
      which( (.data$pm25 < 100) & (.data$mean_diff > 20) ),
      NA)
    )
  
  if ( !returnAllColumns ) {
    hourlyData <- 
      hourlyData %>%
      dplyr::select(.data$datetime, .data$pm25)
  }
  
  # ----- Return ---------------------------------------------------------------
  
  return(hourlyData)
  
}

jonathancallahan, 2019-11-04:
The intent of the chunk of code you highlight is to only allow a `mean_diff > 20` if `pm25 > 100`. The `pm25` variable is the mean.  To accomplish this we replace `pm25` with `NA` whenever `(pm25 < 100) & (mean_diff > 20)`.

It all makes sense to me. If you feel there is a mismatch somewhere could you try again to explain to me what it is?