MazamaScience/AirSensor #221: Trouble loading pat objects

acollier-oxandale, 2020-05-21:
I updated to the latest version of the AirSensor package (version 0.8.0 on the 'master' branch) and I am having trouble loading pat objects. In the examples below, I chose the first sensor and time frame below b/c it is listed as the example in the documentation, for the others I have confirmed that there is data available for the time frames shown using the DataViewer tool. I have also confirmed that each sensor is listed in the pas object. 

Please let me know if you have any idea what could be causing this, and if there are changes I need to make on my end. Thanks!

I've copied the code below and the code with the error messages I'm seeing:

### code

library(AirSensor)
setArchiveBaseUrl("https://airfire-data-exports.s3-us-west-2.amazonaws.com/PurpleAir/v1")

pas <- pas_load()

pat <- pat_load("SCNP_20", 20190411, 20190521)
pat <- pat_loadLatest("SCNP_20")
pat <- pat_loadMonth("SCNP_20")

pat <- pat_load("SCAP_50", 20200425, 20200505)

id <- pas_getDeviceDeploymentIDs(pas, "SCAP_50")
pat <- pat_load(id = id, pas = pas, "SCAP_50", 20200425, 20200505)
pat <- pat_loadLatest("SCAP_50")
pat <- pat_loadMonth("SCAP_50")

pat <- pat_load(pas = pas, label = "SCSB_05", 20200411, 20200515)
pat <- pat_load(label = "SCSB_05", pas = pas, startdate = 20200501, enddate = 20200515, timezone = "America/Los_Angeles")
pat <- pat_loadLatest("SCSB_05")
pat <- pat_loadMonth("SCSB_05")

### error messages 

> pat <- pat_load("SCNP_20", 20190411, 20190521)
Error in pat_distinct(pat) : Not a valid 'pat' object.

> pat <- pat_loadLatest("SCNP_20")
Error in pat_distinct(pat) : Not a valid 'pat' object.

> pat <- pat_loadMonth("SCNP_20")
Error: Data file could not be loaded from: https://airfire-data-exports.s3-us-west-2.amazonaws.com/PurpleAir/v1

> pat <- pat_load("SCAP_50", 20200425, 20200505)
Error in pat_distinct(pat) : Not a valid 'pat' object.

> id <- pas_getDeviceDeploymentIDs(pas, "SCAP_50")
> pat <- pat_load(id = id, pas = pas, "SCAP_50", 20200425, 20200505)
Error in metaList[[length(metaList)]] : 
  attempt to select less than one element in integerOneIndex

> pat <- pat_loadLatest("SCAP_50")
Error in pat_distinct(pat) : Not a valid 'pat' object.

> pat <- pat_loadMonth("SCAP_50")
Error: Data file could not be loaded from: https://airfire-data-exports.s3-us-west-2.amazonaws.com/PurpleAir/v1

> pat <- pat_load(pas = pas, label = "SCSB_05", 20200411, 20200515)
Error in metaList[[length(metaList)]] : 
  attempt to select less than one element in integerOneIndex

> pat <- pat_load(label = "SCSB_05", pas = pas, startdate = 20200501, enddate = 20200515, timezone = "America/Los_Angeles")
Error in metaList[[length(metaList)]] : 
  attempt to select less than one element in integerOneIndex

> pat <- pat_loadLatest("SCSB_05")
Error in pat_distinct(pat) : Not a valid 'pat' object.

> pat <- pat_loadMonth("SCSB_05")
Error: Data file could not be loaded from: https://airfire-data-exports.s3-us-west-2.amazonaws.com/PurpleAir/v1

jonathancallahan, 2020-05-21:
Ashley,

*AirSensor* version 0.8 is incompatible with the current archive available
at the USFS.

This is expected. The new data downloads from ThingSpeak include more
fields of data than we had previously and *AirSensor* 0.8.x now demands
that all fields be present.

Going forward, the following will happen:

   - soon -- Add a `pat_upgrade()` function to upgrade old pat objects by
   adding columns with all NA values. Almost everything will still work.
   - soon -- Rebuild the Mazama archives of South Coast pat objects with
   the new data model.
   - later -- Rebuild the USFS archives of US pat objects.

I hope the clarifies the current situation.

Regards,
Jon

acollier-oxandale, 2020-05-21:
Jon,

Thank you for the clarification - that makes perfect sense.  I'll work with pat_createNew() for now. 

Best,

-Ashley

closed 2020-05-27 by acollier-oxandale