MazamaScience/AirSensor #222: error occurring when converting pat objects to sensor objects

acollier-oxandale, 2020-05-27:
Hi, I'm not sure if this is an issue or an update that I missed related to implementing the functions. I have been seeing the error below occur with many pat objects, I've added one specific example. Is this an issue with the code or is there any way I can troubleshoot this error?

I have updated to the latest version of the AirSensor package (0.8.2), though I was seeing this error before as well. Thank you for the update yesterday, by the way, it's always exciting to see the new iterations of this work. 

And thanks in advance for your help!

____________________________________________________

sensor <- pat_createAirSensor(pat = pat, FUN = PurpleAirQC_hourly_AB_00)

Error in pat_createAirSensor(pat = pat, FUN = PurpleAirQC_hourly_AB_00) : 
  **Error: 'FUN(pat)' does not return regular hourly datetime axis. Please check 'FUN' and see ?pat_createAirSensor for details.**

_____________________________________________________

library(AirSensor)    
setArchiveBaseUrl("http://data.mazamascience.com/PurpleAir/v1")
pas <- pas_load()
pat <- pat_createNew(label = "SCTV_39", pas = pas,  startdate = 20191201, enddate = 20200520) 
pat_multiPlot(pat)

sensor <- pat_createAirSensor(pat = pat, FUN = PurpleAirQC_hourly_AB_00)

____________________________________________________

hansmrtn, 2020-06-11:
This should be resolved in the next update! Thanks for the report @acollier-oxandale!

closed 2020-06-11 by hansmrtn

acollier-oxandale, 2020-06-12:
Thanks for the help and the update Hans!