MazamaScience/AirSensor #236: possible issue with pollution rose and polar plots

acollier-oxandale, 2020-08-20:
I am consistently seeing the following error when trying to generate both pollution rose and polar plots. 

Error in worldmet::importNOAA(code = siteCode, year = year, parallel = FALSE,  : 
  unused argument (parallel = FALSE)
In addition: Warning messages:
1: Unknown or uninitialised column: `USAF`. 
2: Unknown or uninitialised column: `WBAN`. 

This is using version 0.9.9 of the AirSensor package. Is this an issue with the package or are there changes related to the usage of these plots that I may have missed? thanks

_____________________________________________________________________________________
#example code 

library(AirSensor)
setArchiveBaseUrl("http://data.mazamascience.com/PurpleAir/v1/")
pas <- pas_load()

pat <- pat_load(id = id, pas = pas, label = "SCAP_50", startdate = 20200701, enddate = 20200715, timezone = "America/Los_Angeles")
sensor <- pat_createAirSensor(pat = pat, FUN = PurpleAirQC_hourly_AB_00)
sensor_polarPlot(sensor)
sensor_pollutionRose(sensor)

pat <- pat_load(id = id, pas = pas, label = "SCUV_06", startdate = 20200701, enddate = 20200715, timezone = "America/Los_Angeles")
sensor <- pat_createAirSensor(pat = pat, FUN = PurpleAirQC_hourly_AB_01)
sensor_polarPlot(sensor)
sensor_pollutionRose(sensor)

pat <- pat_loadMonth(label = "SCBB_02", pas = pas, datestamp = 202005)
sensor <- pat_createAirSensor(pat)
sensor_polarPlot(sensor, resolution = "normal")

jonathancallahan, 2020-08-20:
Our code was built around worldmet 0.8.8. But I see  at https://github.com/davidcarslaw/worldmet/blob/master/NEWS.md that version 0.9.0 has "Significant changes due to NOAA storage formats and different storage locations".

We will have to update our code to accommodate his changes which are in turn accommodating changes at NOAA.

Relying on open source packages means we are building tools to work with constantly moving targets. This is a big part of what makes building software with "do everything" task lists difficult to support long term.

acollier-oxandale, 2020-08-21:
Thank you for looking into it and for the explanation.

jonathancallahan, 2020-09-13:
Resolved in AirSensor 0.9.16.

closed 2020-09-13 by jonathancallahan