MazamaScience/AirSensor #274: Missing "RData" file for 2022 in the AirFire Monitoring Database

CarlMalings, 2022-02-28:
Hello,

I received the following error when using the AirSensors package to access regulatory monitor data for comparison via the PWFSLSmoke::monitor_load and PWFSLSmoke::monitor_extractData functions:

> Error in MazamaCoreUtils::loadDataFile(filename, baseUrl, dataDir) : 
>   data file could not be loaded from: https://haze.airfire.org/monitoring/AirNow/RData/2022/airnow_PM2.5_2022.RData
> In addition: Warning message:
> In load(conn) :
>   cannot open URL 'https://haze.airfire.org/monitoring/AirNow/RData/2022/airnow_PM2.5_2022.RData': HTTP status was '404 Not Found'

Basically, the "RData" file it is looking for does not exist. It may have just not been created for this year yet, or maybe the way the data are being archived is changing (there is an RData file for this month in the associated directory). I emailed Sim Larkin, listed as the contact for the database here: https://info.airfire.org/monitoring

I also wanted to make you aware of this issue; if the data are being archived in a different way going forward, the package may need to be updated to reflect this. Overall, this package has been very useful and helpful to me in allow an easy platform to compile and analyze multiple sources of air quality data together. Thanks!

jonathancallahan, 2022-02-28:
Hi Carl,

In order to support the growing community of users of this package, I am asking that people post questions to a dedicated package-support slack channel.  I just sent a Slack invite to: carl.a.malings@nasa.gov

Or you should be able to use the following link to join:

https://join.slack.com/t/mazamapackagesupport/shared_invite/zt-14r5zss28-V4w9L028ewKz8yFiXAmjrQ

Please repost your question in the #pwfsl-smoke channel

HaoYinV, 2022-04-09:
> Hi Carl,
> 
> In order to support the growing community of users of this package, I am asking that people post questions to a dedicated package-support slack channel. I just sent a Slack invite to: [carl.a.malings@nasa.gov](mailto:carl.a.malings@nasa.gov)
> 
> Or you should be able to use the following link to join:
> 
> https://join.slack.com/t/mazamapackagesupport/shared_invite/zt-14r5zss28-V4w9L028ewKz8yFiXAmjrQ
> 
> Please repost your question in the #pwfsl-smoke channel

Hi Jonathan,
I also have a question about AirSensor package. May I join the slack group? The link you shared above does not work on my side. My email is yinhao6@gmail.com. Thanks a lot!
Best,
Hao

closed 2023-03-30 by jonathancallahan